rowr111/cubegarden #1: can't clone any of the submodules - permissions issue

rowr111, 2019-04-02:
I showed the error I'm getting to bunnie and he said it's bc the reference to the modules needs to be the read-only https version.

--------------------------------------------
Cloning into 'C:/Users/rowr/documents/GitHub/cubegarden/ChibiOS'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:bunnie/ChibiOS' into submodule path 'C:/Users/rowr/documents/GitHub/cubegarden/ChibiOS' failed
Failed to clone 'ChibiOS'. Retry scheduled
Cloning into 'C:/Users/rowr/documents/GitHub/cubegarden/ChibiOS-Contrib'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:bunnie/ChibiOS-Contrib' into submodule path 'C:/Users/rowr/documents/GitHub/cubegarden/ChibiOS-Contrib' failed
Failed to clone 'ChibiOS-Contrib'. Retry scheduled
Cloning into 'C:/Users/rowr/documents/GitHub/cubegarden/ugfx'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:bunnie/ugfx-xz' into submodule path 'C:/Users/rowr/documents/GitHub/cubegarden/ugfx' failed
Failed to clone 'ugfx'. Retry scheduled
Cloning into 'C:/Users/rowr/documents/GitHub/cubegarden/ChibiOS'...
Host key verification failed.
fatal: Could not read from remote repository.

bunnie, 2019-04-02:
I /think/ this commit fixes it:

139f5ff7ee5747c068389c6c6c52f32047256591

Try pulling the .gitmodules file and then cloning the submodules again?

bunnie, 2019-04-02:
eventually we might want to look into moving these subdirs into rowr111 if we end up doing a lot of work in them but we can also just maintain a branch there for this project.

It's a little tidier of everything lives in this repo tho, I think? less confusing for new people trying to clone stuff and we can get rid of all of the bad habits I had built into those git repos.

rowr111, 2019-04-02:
ok, updating that file fixed the issue, although I had to manually update the repo paths in .git/config to get it to work - I ran github submodule init and I thought that might update .git/config but it didn't.  I don't think anyone else should run into that particular snafu, though.

closed 2019-04-02 by rowr111