rowr111/cubegarden #13: battery testing

rowr111, 2019-05-14:
We need to do various tests related to the battery to find out how long things take/other info. 

Details of these tests to be outlined by @bunnie

bunnie, 2019-05-23:
I think we're starting to get close on battery testing -- I'm going to fully charge and discharge the battery another couple cycles but so far everything is looking about right-on in terms of estimates and lifetime.

bunnie, 2019-05-24:
Some charging notes:

- Empty to 90% in about 2.5 hours
- Empty to 100% in about 5 hours

The last 10% takes a long time because the current draw of the LEDs/electronics subtracts from the top-up current on the battery pack.

bunnie, 2019-05-25:
More notes:

From 100% on a 65% full white pattern:

6h20m to threshold 1
7h15m to threshold 2
7h30m to threshold 3
8h00m to safety
8h20m to shutdown

Recharge:
Red safety pattern
Full charge in 5h10m

Conclusion: need to adjust the brightness thresholds to a higher level, to extend battery life a bit longer on pathologically bright patterns.

bunnie, 2019-05-26:
Adjustments then more testing:

2h15m to threshold 1
4h55m to threshold 2
10h00m to threshold 3
18h35m to safety

This seems to err in the right direction, especially if a cube happens to have a weak battery.

bunnie, 2019-05-27:
26h25m to shutdown

bunnie, 2019-05-27:
Full charge at 4h45m using saftey light pattern (near best case) as the background pattern.

bunnie, 2019-05-29:
Powered off leakage characterization:
4.1455V battery voltage, measured 6PM May 28.
3.9034V battery voltage, measured 5PM May 29.

This is higher leakage than expected. Opened  issue #29 to track and resolve before closing this issue.

bunnie, 2019-06-01:
OK, leakage is now under control. After resolving issue #29, 16 hours on, the battery has gone down by 1.7mV only, which is about where it should be. This is a reasonable rate of decline, comparable to the self-discharge rate of the battery which means now the circuit can probably survive a few months in storage.

It's still strongly recommended to disconnect the batteries at the end of the burn for storage, this will greatly improve the odds that the units can work again one year later.

closed 2019-06-01 by bunnie