rowr111/cubegarden #14: create branch with battery code that is compatible with heart badges

rowr111, 2019-05-14:
We only have a few dev boards, so for all other people that want to do some dev work before we get production boards, and for our 'master badge' concept, we would like to have a branch of code that will continue to function on the heart badges.  The current code has some battery-specific changes that may not be compatible with the old heart badges.

bunnie, 2019-05-16:
Branch is here:

https://github.com/rowr111/cubegarden/tree/heartbadge

I've retained the "app-*" UI structure, graphics libraries, OLED drivers and touch surfaces, so that devs could in theory make a UI for the master cube controller.

There's about 20k of code space and 6k of RAM available, which /should/ be plenty, but if more is needed let's talk. Note that every peer that pings a badge or cube will consume a couple dozen bytes of RAM, and that is dynamically allocated and released based on the peer's visibility, so you can't fill the RAM up entirely.

I'll re-assign to Johan for validation & any follow-up questions.

bunnie, 2019-05-22:
Looks like this is working, so, I'm closing it.

closed 2019-05-22 by bunnie