rowr111/cubegarden #16: battery charging never terminates

rowr111, 2019-05-21:
When plugged in to charge, charging never appears to terminate.  Even when shift is set to like.. 6 (something so high it looks like LEDs aren't even on) charging status is always 'fast charge' and never achieves 'charged'. SoC continually alternates between 99 and 100%.

bunnie, 2019-05-22:
charge terminates on my board now, can you give this a check on yours and close?

need to leave the auto-dimming on for this to work...let's keep that feature in for now, so we can use the boards in a way that's more routine compared to what will happen on playa

rowr111, 2019-05-23:
it reaches charged now, although it took mine about an hour to get from 96% and 'fast charge' state to 99% and charged.  auto-dimming is also working.

closed 2019-05-23 by rowr111