rowr111/cubegarden #25: master badge needs to be able to respond to sex requests even when lightgene is not active

rowr111, 2019-05-28:
It should use an extreme effect when responding to requests so that people can get a wacky result when interacting with the cubegarden.

rowr111, 2019-07-01:
master badge and cubes only have one lightgene effect - it will now respond to sex requests hardcoded to that lightgene with commit a016c17

bunnie, 2019-07-29:
an extreme "default" effect has been added. I think you can still do something stupid like have sex with someone and dilute out the extreme effect but at least if you initialize the badge from a clean image, it loads the most extreme/rare parameters i can think of.

Net result is, it looks like a clown car. 

Closed for now, open if you think you want more done.

closed 2019-07-29 by bunnie