rowr111/cubegarden #35: Connect to Altar

drakeley, 2019-06-17:
Hello!

I will paste more info after I talk with Shlomo, but it would be awesome if we could coordinate so that the cubes can also respond to the button that the DJ will use to light up the wedding altar! I am happy to do whatever needs to be done to make this happen!

rowr111, 2019-06-18:
The cubes respond to radio commands that can trigger a change in the current permanent effect or trigger a temporary effect - it's possible for you program something that could send the appropriate commands.. join our tech chat group and let's talk more about what you want to do!

drakeley, 2019-06-26:
Closing as "won't fix" as the cubes already have a "beat drop detection mechanism" in DB mode!

closed 2019-06-26 by drakeley