rowr111/cubegarden #36: improve communication received rate between master badge/cubes

rowr111, 2019-06-19:
It is currently not expected that 100% of the cubes will receive messages from the master badge, for various (not all known) reasons.  It would be good to improve this so that more of the cubes will pick up changes without having to re-send from the master badge.

drakeley, 2019-06-19:
I'm super interested in taking this on - I know there are lots of ways we can do this, so I'll do a bit of research and come up with a proposal that I'll post here

boberj, 2019-06-20:
I looked into this a while back and here are some resources that could be useful:
[Adafruit's guide on RFM69HCW](https://learn.adafruit.com/adafruit-feather-32u4-radio-with-rfm69hcw-module/using-the-rfm69-radio) which points to [LowPowerLab's Arduino library](https://github.com/lowpowerlab/rfm69) and the [Radiohead packet radio library](http://www.airspayce.com/mikem/arduino/RadioHead/).

[Overview of the RFM69 chip](https://learn.sparkfun.com/tutorials/rfm69hcw-hookup-guide/all)

One thing to keep in mind is that we want to be backwards compatible with the existing heart badges so we can't use one of the above libraries straight out of the box. It might however be possible to modify the code to function as a "sub-protocol" in our existing network setup, or simply use them as inspiration to implement our own reliable transport.

Alternatively, if it turns out that nobody interacts with the cubes using the heart badges we could consider switching out the network stack for next year.

The Radiohead library comes with multiple networking modes and it would be cool to implement a mesh network but maybe not necessary given how we use the radio.

The way we currently use it is that we broadcast all commands (except address assignment). For broadcasts there are no ACKs so we would have to send the messages individually to all nodes if we wanted reliability. By instead retransmitting the broadcasted message multiple times we might get almost the same amount of reliability as one of the reliable transports.

bunnie, 2019-08-05:
I presume this is not going to happen before the burn. Removing pre-playa tag.