rowr111/cubegarden #39: board occasionally crashes upon startup after code push (pre-test sequence)

rowr111, 2019-06-30:
serial output is:
Cubegarden bootloader.  Based on build c29b27890027a66d02501b3c4b81514edfd9fa7c
Core free memory

over and over again (every 5-10 sec reported - watchdog time is restarting it but it keeps crashing) - it never gets to the test sequence. 
Red light of doom comes on.  
Unplugging both power and battery and then replugging in battery fixes the problem.
Happens when rpi is plugged in and a build is pushed to the cube board.
Only has happened so far to the initial prototype boards, but has happened to two of them (Jeanie's and Johan's, same exact symptoms reported).  Has not happened when starting/restarting otherwise.

bunnie, 2019-07-29:
I haven't seen this one happen recently...any more incidences?

rowr111, 2019-07-30:
I haven't seen it in a while.  Since there is a workaround and it doesn't seem to affect the actual functioning of the board, I'm fine with leaving this as a won't fix.

closed 2019-07-30 by rowr111