rowr111/cubegarden #57: cubemaster name doesn't show up in badge UI or on other people's badges

rowr111, 2019-08-07:
I see that 'cubemaster' is being set as the name in genes.c of the master badge but I never see that name displayed anywhere in the UI. In my test board I only see the old name that was set back when it was a bm 17 board.  That same old name also appears in other people's badges when they are looking for friends.

bunnie, 2019-08-08:
I think you need to completely wipe your cubemaster board, the lightgene record is different from userconfig, and that would be retained from previous configurations. 

To do this, do the following.

1. Make sure openocd is running.

2. From the src-heart directory, telnet to the openocd session with `telnet localhost 4444`

You'll see something like

```
Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'
Open On-Chip Debugger
>
```
3.  At the prompt, type `kinetis mdm mass_erase`. This will wipe the *entire* badge clean, including all of its previous memories of who it was. 

4. Now issue:
```
program build/cubemaster.elf
kinetis mdm reset
```

5. At this point the badge should reset and will come out saying it's the cubemaster, with all defaults.

bunnie, 2019-08-08:
Also, type `exit` to quit the telnet session

bunnie, 2019-08-08:
Another "cheesy" way to do this, instead of doing the full erase (but I recommend you do it because it really guarantees a clean slate), is to bump the "GENE_VERSION" record up by one in genes.h

But if you do this, there could still be other records I forgot to wipe. Usually, the assumption is you don't want the badge to forget its settings or who it was whenever you update the code, so the whole process is structured to conserve as much of that as possible.

bunnie, 2019-08-12:
is this one working now?

rowr111, 2019-08-12:
Yes!  working.  Forgot to close this after build day.  all good.  Thank you for the instructions.

closed 2019-08-12 by rowr111