rowr111/cubegarden #66: investigate what master badge 2.0 might look like/do

rowr111, 2019-09-04:
Is it a remote control with buttons?  Is it something with a screen and menu options?
I feel like this is worth talking about.. our current master badge has maxed out its code capacity - we need to decide if we are going to make the cubes function as a group independently of the master and just use some basic interface for adjusting parameters, or what.

boberj, 2019-09-09:
My thinking on this is to use an Arduino connected to a phone for UI. I did even buy two before the burn to experiment with but didn't have time to finish that in time.

The Arduino could run system managing tasks on its own. By plugging it into a phone one would get a nice UI to control effects and settings.

rowr111, 2020-07-14:
after some conversation I think we have agreed to retain the master badge for basic functionality/control but @boberj will also work simultaneously on an arduino based solution.

closed 2020-07-14 by rowr111