rowr111/cubegarden #67: add dB effect as an optional overlay on effects instead of its own effect

rowr111, 2019-09-04:
Basically this effect was so popular that it seems integral to the cubegarden experience.  We should just add it on top of effects.

boberj, 2019-09-09:
I want to simplify effects development by creating a way to more abstractly describe effects (color change, pulsing etc.) and then layer/blend them together to create a complete new effect. Seems like that could be used to implement this by treating a complete effect as just another layer.

rowr111, 2020-07-14:
yeah.. my general idea is along those lines - the base effect would reduced to be color palettes only.  you'd pick the color palette you'd like, and then build your effect by picking all the other 'moving' parts - pulsing, db effect, tilting, change color on tap, etc.

rowr111, 2020-07-24:
this effect is in and looking pretty good (imho, lol) as of commit 4d1a8d0

closed 2020-07-24 by rowr111